sipeed/NanoKVM #577: Keyboard and Mouse Inputs Not Fully Redirected to Remote System

hafeeshf, 2025-07-12:
I'm using a remote system through a browser (e.g., via NanoKVM ), and I'm facing an issue where keyboard shortcuts and mouse inputs are not being fully redirected to the remote machine.

For example:

When I make the remote screen fullscreen on my local computer and press the Esc key, it exits fullscreen on the local system, instead of sending the Esc key to the remote system.

Similarly, when I trigger my AutoHotkey scripts using keyboard shortcuts (like Ctrl+Alt+X), they are executed on the local machine, not the remote one.

Some other keyboard shortcuts (like Alt+Tab, Ctrl+Win+Arrow, etc.) are also not reaching the remote system and are instead acting locally.

Feature Request:
Please provide an option to fully bind and redirect the local keyboard and mouse to the remote session, so that:

All inputs from my local devices are sent directly to the remote system.

No local shortcuts interfere with remote usage.

AutoHotkey or other automation tools (running on the remote system) can be triggered seamlessly.

This would greatly improve usability, especially for power users relying on automation or performing remote troubleshooting.

aarcro, 2025-10-30:
> > Same issue?
> > 
> > * [Lock the keyboard in fullscreen mode #575](https://github.com/sipeed/NanoKVM/issues/575)
> 
> My request is simpler than this one. Even with the `keyboard.lock()` in place not all the keystrokes pass through. I particularly have issues with `alt-tab`. It will fire once, but not continue to hold `alt` down. This means I can toggle two windows, but not cycle all open windows.

If anyone else has that issue - I turned on sticky keys in windows on the target machine. I can double tap alt to hold it down, then hit tab to cycle through, and alt again to release.

kha84, 2025-10-31:
Smart workaround @aarcro , thanks for sharing!
I wonder if there's a way to use a standard VNC or alike client with NanoKVM instead of using a web-based VNC client? I guess we wouldn't have all these questions, if the device was provided just a regular VNC interface