sipeed/NanoKVM #593: Keyboard Input Not Working After Update from 2.2.6 to 2.2.9

horkryzard-home, 2025-07-29:
Hello,

I have installed and used the NanoKVM PCI with firmware version 2.2.6 without any issues. However, after upgrading to version 2.2.9, I lost the ability to send keyboard input.

The NanoKVM PCI is connected via USB. I have tried multiple USB ports.
Once my server is fully booted, restarting the NanoKVM PCI temporarily restores keyboard functionality - but it stops working again after the next server reboot. This prevents me from accessing the BIOS, which is crucial because the server still requires a few BIOS tweaks.

Is there any option to downgrade back to version 2.2.6?

Thank you in advance.

myde2001, 2025-08-21:
still happening, very annoying

Lexden12, 2025-09-07:
Similar issue with mine:
`
[    5.049767] usb 1-12.3: new high-speed USB device number 14 using xhci_hcd
[   10.402904] usb 1-12.3: device descriptor read/64, error -110
[   26.263915] usb 1-12.3: device descriptor read/64, error -110
[   26.432764] usb 1-12.3: new high-speed USB device number 15 using xhci_hcd
[   31.895914] usb 1-12.3: device descriptor read/64, error -110
[   47.778912] usb 1-12.3: device descriptor read/64, error -110
[   47.882329] usb 1-12-port3: attempt power cycle
[   48.756762] usb 1-12.3: new high-speed USB device number 19 using xhci_hcd
[   53.849918] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   59.481863] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   59.689756] usb 1-12.3: device not accepting address 19, error -62
[   59.752760] usb 1-12.3: new high-speed USB device number 20 using xhci_hcd
[   65.113857] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   70.745855] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   70.953755] usb 1-12.3: device not accepting address 20, error -62
[   70.954246] usb 1-12-port3: unable to enumerate USB device`

Rebooting the NanoKVM fixes it, but it is certainly frustrating to sometimes get stuck with my PC taking 70 seconds to boot because the NanoKVM's USB gadget broke.

Pyroboy5000, 2025-09-27:
i have this issue as well. no keyboard or mouse input after the update. i swapped the kvm out for a new one with older firmware and its working on the system. how can i fix this one with the newer firmware?