sipeed/NanoKVM #594: No video from notebook or laptop till OS boots.

Trogvars, 2025-07-30:
Can someone else prove that NanoKVM doesn't receive signal from notebook/laptop HDMI until OS fully boot?

Latest firmware 1.4.0, Full version. Tried with 5 or 6 different laptops, different cables and settings.
No video until system boots up and catch it like a second monitor. 

I guess this is not a NanoKVM issue but laptops architecture

 Everything is working fine with desktops. It shows the picture right after power is on.

Issue only seen on laptops.

jamesfreeman959, 2025-08-02:
Hey there. This seems to be an inherent limitation of the NanoKVM - it's not just laptops - there's a lot of system that it just doesn't work with, especially in its current form. I can't work out whether it's a hardware or software limitation - but it doesn't like a lot of systems especially with UEFI/BIOS screens, and also Linux/*BSD in text mode - all of the above it really struggles with. I have had some success by putting an EDID emulator in line between the HDMI out of the device I'm monitoring, and the NanoKVM, but this fixed some things and introduced other problems. I've tried a few things with recompiling the code too, but for example there's a block in the code that sets the EDID of the HDMI capture. After a few efforts at building different EDID's with no change, I finally realised that there are entire functions in the kvm_vision code that are just not being used - so perhaps they are for "future implementation". 

I've tried to document my efforts here in case it helps anyone else: https://github.com/sipeed/NanoKVM/issues/571

In short I've run out of time to spend on this right now as I haven't made any progress. It's a real shame - the whole purpose of a device like this is to see the screen before the OS boots - by that stage you have remote desktop capabilities normally. There's very little response to posts here or on the forum - if I get any ideas I'll follow it up but right now I'm stumped, and I don't have any more time to spend on this.

AT-StephenDetomasi, 2025-08-22:
I'm just chiming in to say that I'm having the exact same issue. Super frustrating as the whole reason I bought this was to access the BIOS remotely. Very annoying!