sipeed/NanoKVM #612: Shift button issue

doesntremindm3, 2025-08-26:
I have about 12 of these I use.  Over time, they seem to develop a very strange problem where the Shift key does not work correctly.  Shift only starts "taking effect" on the second key press when it is held down.  For example, if I try to type 

`EXAMPLE`

I see 

`eXAMPLE`

Another example is if I try to type 

`!@#$%`

I see

`1@#$%`

Sometimes, I can solve the issue by reflashing the SD card.  Other times, this does not seem to work.  Has anybody seen this or do you have any advice?  I have seen the issue with some of my Cube and PCIe ones.

Alvalvar, 2025-09-15:
I have the same problem :(

Alvalvar, 2025-09-19:
I found a workaround: every time you need to enter uppercase characters in sequence (with the shift key), be sure to release the shift key before each press.

Brains1959, 2025-11-07:
Same here. Just installed a NanoKVM PCIe (Image v1.4.1, Application v2.2.9) and I have the same issue. The first character of a shifted sequence is non-shifted. Pretty tricky entering a password with Upper, Lower and Special characters. And the shift key of the on-screen keyboard does not work at all (because every key is the first one after pressing shift I presume). Entering characters using the Paste function appears to be working as expected. Tried Chrome and Edge.

Brains1959, 2025-11-07:
Additional information: The same appears to happen with the Control (Ctrl) key. First character of a sequence is like the Control key is not pressed yet.
A (very clunky) workaround (until this is fixed) is to use a non-functional key as the first press after shift (e.g. right arrow in most cases). Or press any key followed by back-space and then the desired key while holding the shift key down.