sipeed/NanoKVM #617: USB serial presented to device

kjonca, 2025-08-28:
I do not like serial number like '0123456789ABCDEF' 
i can see this is set in /etc/init.d/S03usbdev (around line 19)
Can we set this number to somethind more unique? 
board HW id (if exists)

dormancygrace, 2026-09-14:
Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

The USB serial number is derived from the SoC UID in both gadget profiles, with a fallback only when no usable UID is available. This carries forward the idea in PR #746 with UID validation.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/kvmapp/system/init.d/S03usbdev#L88).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!

kjonca, 2026-09-25:
> Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.
>
> The USB serial number is [...]
This carries forward the idea in PR [#746](https://github.com/sipeed/NanoKVM/pull/746) with UID validation.

I have put something similar on my kvm, but is overwritten during upgrade.
But if two people invented something similar it must be good, it isn't? :)